hoggyhoggy/givenergy-modbus-async #9: invalid TransparentResponse frames

divenal, 2024-04-06:
https://github.com/hoggyhoggy/givenergy-modbus-async/blob/85675ba2a0c03a4a6dd0f7f81d1d3cd65db2716b/givenergy_modbus/pdu/transparent.py#L152

This probably ought to be a subclass of ExceptionBase, or perhaps InvalidPduState, rather than using a core exception. Not sure why, but I'm frequently getting one of these thrown with code #57 when I run my snoop code. I had taken it to be a one-off random corruption, but it seems to be happening too often for that.

eg

```
INFO:__main__:Processing 2:4/ReadInputRegistersResponse(slave_address=0x01 base_register=60)
INFO:__main__:Processing 2:4/ReadInputRegistersResponse(slave_address=0x11 base_register=0)
INFO:__main__:Processing 2:4/ReadInputRegistersResponse(slave_address=0x01 base_register=60)
TransparentResponse function #57 decoder
INFO:__main__:Processing 2:4/ReadInputRegistersResponse(slave_address=0x30 base_register=0)
INFO:__main__:Processing 2:4/ReadInputRegistersResponse(slave_address=0x30 base_register=0)
...
INFO:__main__:Processing 2:4/ReadInputRegistersResponse(slave_address=0x36 base_register=60)
INFO:__main__:Processing 2:4/ReadInputRegistersResponse(slave_address=0x37 base_register=60)
TransparentResponse function #57 decoder
INFO:__main__:Processing 1/HeartbeatRequest(data_adapter_serial_number=WH2327G406 data_adapter_type=1)
TransparentResponse function #57 decoder

```

The snoop code in question is

```
import asyncio
import logging
import socket
import sys
from typing import Callable, Dict, List, Optional, Tuple

from givenergy_modbus.exceptions import CommunicationError, ExceptionBase
from givenergy_modbus.framer import ClientFramer

_logger = logging.getLogger(__name__)

async def snoop():
    """Passively watch modbus traffic."""

    connected = False
    reader: StreamReader
    network_consumer_task: Task
    connect_timeout = 2.0

    framer = ClientFramer()

    host=sys.argv[1]
    port=8899
    try:
        connection = asyncio.open_connection(host=sys.argv[1], port=8899, flags=socket.TCP_NODELAY)
        reader, writer = await asyncio.wait_for(connection, timeout=connect_timeout)
    except OSError as e:
        raise CommunicationError(f'Error connecting to {host}:{port}') from e

    while reader and not reader.at_eof():
        frame = await reader.read(300)
        try:
            async for message in framer.decode(frame):
                _logger.info('Processing %s', message)
        except (ExceptionBase, NotImplementedError) as e:
            print(e)
    _logger.critical('network_consumer reader at EOF, cannot continue')

if __name__ == "__main__":
    logging.basicConfig(level=logging.INFO)
    asyncio.run(snoop())

```

divenal, 2024-04-07:
Looking closer at the bad message... after the serial number and the padding sequence, I get the byte sequence
 31 39 32 2e 31 36 38 2e 31 2e 39 34 2c 32 35 35 2e 32 35 35 2e 32 35 35 2e 30 2c 31 39 32 2e 31 36 38 2e 31 2e 32 35 34 0d 0a 0d 0a 53 93
which is immediately obviously ascii, and is the string
  "192.168.1.94,255.255.255.0,192.168.1.254\r\n\r\n"
presumably followed by the checksum.

And that triplet is (ip address, netmask, gateway) for my inverter. Which is a very odd thing for the dongle to be spitting out.   I don't _think_ it's a feature of the way I'm capturing traffic, but not impossible.

hoggyhoggy, 2024-04-08:
Is it just on the heartbeat initiated ones? I think I've seen this before on another brand but I'll have to look around. 
I believe that it's not just the inverter you can modify settings on & that the dongle too presumably has registers (so you can potentially change all the things that are greyed out on the interface pages & probably more, such as setting polling intervals, firmware update management and so on) remotely although quite how that works and what they are I don't know.

divenal, 2024-04-08:
Yes, I was just starting to think that it might be because I'm not responding to the heartbeat requests - I've only seen it on a passive tcp connection, and not on a proper client connection which responds to heartbeats.

But of course if a heartbeat response gets dropped, or is too late, then it could in principle happen on a real connection.

hoggyhoggy, 2024-04-08:
If your Google fu is good it is perhaps explained in a German Smarthome Forum....

divenal, 2024-04-09:
I saw (probably) the same issue running my monitoring script on the giv_tcp version of givenergy_modbus:

 No decoder for function code 57

hoggyhoggy, 2024-04-10:
It is an odd one. Did you find the thread I was alluding to on Loxforum? It does help explain things I can't.

divenal, 2024-04-10:
Yes, lots of interesting info in there. Does explain that transparent stuff...

divenal, 2024-04-12:
Just had something new turn up.  My inverter's dongle has just had a firmware update, and I'm getting a new error
TransparentResponse function #22 decoder

The frame in question starts in the usual way : 59 59 00 01 00 9e 01 02 then the serial number and the padding
Then 01 16 46 44 32 33 3x 3x 3x 47 3x 3x 3x  (ie more serial number)
 00 3c 00 3c 01 50 ae 49 53 48 46 51 01 15 00 01 00 01 00 01 25 80 00 00 00 and lots more zeros.

Now 0x3c is 60,  so this could be doing something to 60 registers starting at 60.

In previous logs, I've been seeing a lot of ReadInputRegisterResponses from slave address 1 (as here), but obviously not a set of battery registers.  I wonder if they've now given that a new code to distinguish it ?

Obviously I'm only seeing the response - don't know who's asking for it.

Off the top of my head...   could it be a one-packet summary of the key values across all the batteries, to reduce bandwidth ?

divenal, 2024-05-08:
I'm still getting these #22 messages  - my live clip-monitoring system reported some last night (which I forgot to save), and some more this morning:

```
 1112 1070.9 1070.9   1001  958.6  958.6     807   -28  -28.0  1 1 56946 240
 1425 1336.5 1336.5   1284 1202.7 1202.7    1093   -27  -27.2  1 1 57187 223
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
 1217 1246.9 1332.0   1081 1111.4 1198.1     893   -28  -27.8  1 1 57410 228
 1137 1164.5 1323.6   1009 1034.6 1189.9     840   -28  -28.0  1 1 57638 228
```

(The message text is wrong - these are unsolicited messages in addition to expected response.)

Matching up with timestamped logfile, it was around 2024-05-08 08:24:59

divenal, 2024-05-08:
And another set:

```
 2678 2685.2 3297.5   3936 3938.7 3946.3    3672  1600 1593.8  4 1 7558 122
 2424 2489.3 3257.1   3666 3734.2 3935.6    3392  1599 1597.7  4 1 7680 124
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
Expected response never arrived but resulted in exception: TransparentResponse function #22 decoder
 2440 2452.3 3216.9   3716 3720.5 3924.9    3444  1598 1597.9  4 1 7805 126
  962 1334.6 3122.8   2321 2670.9 3862.2    2035  1595 1595.7  4 1 7931 131

```
another group of 8, this time around 2024-05-08 18:26:20.

divenal, 2024-05-10:
Oh... I think I've just stumbled upon this mysterious function #22 thing.  Just happened to be reading pdu/transparent.py and noticed

```
class TransparentRequest(TransparentMessage, ClientOutgoingMessage, ABC):
    """Root of the hierarchy for Transparent Request PDUs."""

    @classmethod
    def lookup_transparent_function_decoder(
        cls, transparent_function_code: int
    ) -> type["TransparentRequest"]:
...
        elif transparent_function_code == 6:
            return WriteHoldingRegisterRequest
        elif transparent_function_code == 0x16:
            return ReadBatteryInputRegistersRequest
```

But then its partner class TransparentResponse has the same function, but manages to omit handling case 0x16. Since our code decodes responses rather than requests, it missed it.

(The pdu code seems ridiculously convoluted - lots of intermediate helper and "mixin" classes that don't really add terribly much. Yet more code that could be dramatically simplified. Not quite sure there's much to be gained from separate Request and Response hierarchies - clearly having them has resulted in this omission.)

Not sure what the difference between ReadInputRegisters and ReadBatteryInputRegisters is. Nor whether all inverters support the latter.

divenal, 2024-05-10:
Just seen a note on the forum about GE polling installations nightly - looking for battery health ?  I wonder if it's them using this reqest. I've been unable to trigger the response when playing around with the apps.

divenal, 2024-05-30:
Just seen some more unexpected packets:
`Expected response never arrived but resulted in exception: No decoder for transparent function #38`

But those coincide with a firmware update, so I'm guessing they're something to do with that.

divenal, 2024-07-23:
And I've also seen input registers read from slave address 1, which now turns out to be meter results.