hoggyhoggy/givenergy-modbus-async #17: Want 1s minimum between requests ?

divenal, 2024-04-13:
https://github.com/hoggyhoggy/givenergy-modbus-async/blob/85675ba2a0c03a4a6dd0f7f81d1d3cd65db2716b/givenergy_modbus/client/client.py#L231

Did we decide that we ought to wait 1s between requests?  (This delay is just between sends - not from receipt of response.)

hoggyhoggy, 2024-04-15:
It was suggested that a min between requests was 1sec. Although it currently doesn't seem to mind from what I've tried when trying to see how quick it can get.

divenal, 2024-04-16:
Yes, it does seem to work fine with faster updates. So it's really just a matter of what it should default to, with a comment that it seems to fine with faster updates if you want to do that.

britkat1980, 2024-04-30:
I’d suggest that 0.5s is fine. 1s would mean an added 7-8s for some models, which have lots more registers to retrieve.