hoggyhoggy/givenergy-modbus-async #30: disabling charge target when limit is 100%

divenal, 2024-04-25:
https://github.com/hoggyhoggy/givenergy-modbus-async/blob/50d72dd3914777f633517b1b3c3541326ed679dd/givenergy_modbus/client/commands.py#L109

Is it actually necessary to imlement  set_charge_target(100) as disable_charge_target(), rather than just leaving the charge target enabled, but with a value of 100% ?

Problem is that newer inverters have multiple charge slots, and you don't want to turn off the flag unless you know that they're _all_ set to 100%.

britkat1980, 2024-05-01:
agreed, I think this can go

hoggyhoggy, 2024-05-02:
I believe this does indeed seem redundant. The setting of the charge target flag itself (used to be called "Winter Mode") is still relevant but I'm 99% sure that whatever the target is actually set to does not matter. (Given you have newer gen kit & I don't, we can confirm this relatively easily.)