hoggyhoggy/givenergy-modbus-async #34: Over-Read/Write Protection

hoggyhoggy, 2024-05-04:
It appears that registers are not held in RAM therefore each read/write imparts "wear" on the systems flash. 
We can't count read/writes reliably but what we can probably do is is add protection such that if an end users script gets stuck in a constant loop we stop sending them to the inverter. 
Actual "what constitutes a 'bad' loop condition to be discussed"

Also need to add some sort of info to the documentation around the impact of over use of the library to similar effect. "This may wear your system controller more than normal...." 

Opens up more of a use case for "listen_plant" such that multiple devices aren't all requesting the same data directly & as such just use the data already being sent across the connection instead.

divenal, 2024-05-04:
I wonder if it might be eeprom rather than flash?  Flash is much higher density (Mb or Gb), but requires erases in blocks (maybe 4k?) rather than at individual byte level.

From https://en.wikipedia.org/wiki/Eeprom:

> [Flash memory](https://en.wikipedia.org/wiki/Flash_memory) is a type of EEPROM designed for high speed and high density, at the expense of large erase blocks (typically 512 bytes or larger) and limited number of write cycles (often 10,000). There is no clear boundary dividing the two, but the term "EEPROM" is generally used to describe non-volatile memory with small erase blocks (as small as one byte) and a long lifetime (typically 1,000,000 cycles).

If we suppose lifetime of 10 years, that's 10,000 per year or 30 per day ?

Could get round block erasure issue by using some sort of circular log structure, I suppose.

Requesting data would be read-only, and shouldn't have any wearing ?

hoggyhoggy, 2024-05-09:
I know the exact one in Gen1 as I have one in bits in the garage. I'll post it when  I get back home late next week. 
Gen2 and 3 have different ones as they are more powerful but can't realistically have worse performing write limits I'd imagine.