hoggyhoggy/givenergy-modbus-async #35: What does read_registers.py is_suspicious do ?

divenal, 2024-05-06:
https://github.com/hoggyhoggy/givenergy-modbus-async/blob/315535a154b04a9c18aa232cd4880aa0220b8bc2/givenergy_modbus/pdu/read_registers.py#L141

this defines a function is_suspicious which looks for particular values in particular slots of any block of registers. Doesn't seem to be called from anywhere. Do we know what it's looking for ?

hoggyhoggy, 2024-05-09:
There is/was a known bunch of corrupted responses that the inverter could respond with. I did t think this was used anymore however I think Britkat has added a new one so I guess one of the new devices still has known issues?

divenal, 2024-07-19:
I did get one rather anomalous status line from my logging a few weeks back, and haven't got round to investigating. (I currently record all the raw binary in a capture file, so that I can do offline analysis if needed.)

Ah - here it is:
```
                 ----------solar----------  --------gen-------    --export--   ---------battery------------   -temp- -time-

2024-06-24: 1418   5973 5977.9 5879.1 5991.7  4972 4972.9 4971.0   4752 4752.2   -601 -608.2 50 15 0 0  42   57.2   2910 30
2024-06-24: 1418   6127 6089.7 5903.9 6089.7  4973 4973.0 4971.2   4752 4752.0   -777 -734.8 50 15 0 0  42   57.2   2940 30
2024-06-24: 1419   5929 5969.2 5906.4 6083.7  4974 4973.7 4971.5  -2962 -1033.5  17376 12848.3 50 15 0 0 39884   4932.0   2970 30
2024-06-24: 1419   1941 2948.0 5509.9 5926.9  3172 3622.4 4791.5   2927 1936.9   1495 4333.3 50 15 0 0  42   57.4   3000 30
2024-06-24: 1420   2308 2468.0 5189.7 5754.0  3246 3340.1 4637.0   3019 2748.5   1492 2202.3 50 15 0 0  42   57.5   3030 30

```

where the middle line is claiming an outrageous battery charge rate, Soc, and inverter temperature. Must go back and see what was in the raw data.  I was wondering about enabling crc checks, but would need to check what gen-1 systems do - while the dongle didn't verify the checksum, presumably it still added one to outgoing packets.  Anyway, something to consider.